crisap94/MHZ19 #3: nothing seems to work

cfozzi, 2019-04-05:
please someone help! NOTHING ABOUT THE EXAMPLE WORKS! First the libary name is wrong because the libary is not called "MHZ19_uart.h" as it is in the example! that does not work! i had to change it to "MHZ19.h" to get it to work! But even then the line that says "MHZ19_uart mhz19;" would give me an error. if i remove that line and fix the libary name it still does not work and it takes me to the line that says "int co2ppm = mhz19.getPPM();" and tells me "'mhz19' was not declared in this scope". someone please help i just spent £20 and waited a month for my sensor just to find that the code does not work!

crisap94, 2019-04-05:
Hey.... Welll... the sensor Works, Calm down, the example is outdated... As i forked the library and implemented new features but the owner did not like it.... i will upload the exaple code soon.

crisap94, 2019-04-05:
Example updated and Readme updated.... se last changes from the library to know how to use it.

closed 2019-04-05 by crisap94

cfozzi, 2019-04-05:
wow thanks for the quick fix! sorry for my first message i was frustrated.

cfozzi, 2019-04-05:
The new code works and is uploaded to my arduino and can interface with the sensor but after 10 minutes of warming the status is still 0. Is my sensor broken? am i missing something? does it take longer to warm up?

Kumalix, 2019-04-07:
I have the same problem. After 1 hour it is still warming up?
Is this a liberary issue? or wiring?

cfozzi, 2019-04-10:
I removed the warming loop and replaced it with a loop that counts down from 100. i have found that 100 seconds is enough time for it to warm up and function properly.

[Mh-Z19_working.zip](https://github.com/crisap94/MHZ19/files/3063650/Mh-Z19_working.zip)