crisap94/MHZ19 #7: hello! I can not work by PWM

HanSeoHyun, 2019-06-03:
hello! I want to use nodeMCU esp8266 12E, and MH_Z19B sensor by PWM
but, It does not work.
help me please...

-------------------------
#include <Arduino.h>
#include "MHZ19.h"

//const int rx_pin = 13; //Serial rx pin no
//const int tx_pin = 15; //Serial tx pin no
const int pwmpin = 14;
//MHZ19 *mhz19_uart = new MHZ19(rx_pin,tx_pin);
MHZ19 *mhz19_pwm = new MHZ19(pwmpin);

/*----------------------------------------------------------
    MH-Z19 CO2 sensor  setup
  ----------------------------------------------------------*/
void setup()
{
    Serial.begin(115200);
    mhz19_pwm -> begin(pwmpin);
   // mhz19_uart->begin(rx_pin, tx_pin);
    mhz19_pwm->setAutoCalibration(false);
    while (mhz19_pwm->isWarming())
    {
        Serial.print("MH-Z19 now warming up...  status:");
        Serial.println(mhz19_pwm->getStatus());
        delay(1000);
    }
}

/*----------------------------------------------------------
    MH-Z19 CO2 sensor  loop
  ----------------------------------------------------------*/
void loop()
{
    int co2ppm = mhz19_pwm->getPPM(MHZ19_POTOCOL::PWM);
    int temp = mhz19_pwm->getTemperature();

    //Serial.print("co2: ");
    //Serial.println(co2ppm);
    //Serial.print("temp: ");
    //Serial.println(temp);

    co2ppm = mhz19_pwm->getPPM(MHZ19_POTOCOL::PWM);
    Serial.print("co2: ");
    Serial.println(co2ppm);
    

    delay(5000);
}

-----------------------------
sensor-----nodeMCU
PWM-----D5
GND-----GND
Vin------Vin
-----------------------------

error: can't get MH-Z19 response.
MH-Z19 now warming up...  status:error: can't get MH-Z19 response.
-1

that messages are show infinite....

-------------------------------
I tested my nodeMCU by another code, And I tested my sensor by UNO. They works well.

How can I work well...? Help me please.....

crisap94, 2019-10-28:
Would you mind retry again with the new example, the library just had a refactor.

closed 2019-10-28 by crisap94

MagTun, 2019-11-16:
I uploaded the [MHZ19_getppm.ino](https://github.com/crisap94/MHZ19/blob/master/examples/MHZ19_getppm/MHZ19_getppm.ino) sketch to my nodeMCU esp8266 and used the same connection as mentionned above:
```
sensor-----nodeMCU
   PWM-----D5
   GND-----GND
   Vin-----Vu
```

and I am still getting the same error:
```
MH-Z19 now warming up...  status:error: can't get MH-Z19 response.
-1
```

Any idea how to solve this?