crisap94/MHZ19 #14: Serial Monitor says the the sensor is warming up...forever.

cbwatts, 2020-08-23:
Not sure what the fix is. If I remove the warm up part, and instead add a 3 minute delay in the setup, it kind of works. The PWM and uart have drastically different values however...any suggestions?

`#include "MHZ19.h"

const int rx_pin = 6; //Serial rx pin no
const int tx_pin = 7; //Serial tx pin no

const int pwmpin = 10;

MHZ19 *mhz19_uart = new MHZ19(rx_pin,tx_pin);
MHZ19 *mhz19_pwm = new MHZ19(pwmpin);

/*----------------------------------------------------------
    MH-Z19 CO2 sensor  setup
  ----------------------------------------------------------*/
void setup()
{
    Serial.begin(115200);
    mhz19_uart->begin(rx_pin, tx_pin);
    mhz19_uart->setAutoCalibration(false);
    delay(180000);
}

/*----------------------------------------------------------
    MH-Z19 CO2 sensor  loop
  ----------------------------------------------------------*/
void loop()
{
    measurement_t m = mhz19_uart->getMeasurement();

    Serial.print("Uart co2: ");
    Serial.println(m.co2_ppm);
    Serial.print("Temp: ");
    Serial.println(m.temperature);

    int co2ppm = mhz19_pwm->getPpmPwm();
    Serial.print("PWM co2: ");
    Serial.println(co2ppm);
    
    delay(5000);
}`

I would love to have the warm up in there...does anyone have any advice? I'm a novice.

crisap94, 2020-12-13:
Seems this is related to issue #12

crisap94, 2020-12-14:
@cbwatts I did change the examples and the documentation for newer users not to get stuck on it. 
Thanks for finding a temporary workaround. 
Feel free to contribute, I'd appreciate some help.